officinerobotiche/uNAVPCB #2: 5 volt voltage regulator on board, why not?

bobboteck, 2015-01-12:
I saw that you have removed the 5Volt voltage regulator, this is not a choice that I like. Personally, I usually use a switching regulator Output 6.5 / 7.0V, this voltage reaches the various boards, where each has its voltage regulator for the required voltage. This, in my view, be also another benefit with regard to the cleaning of the supply signal, a cable that carries the 5V around for the robot is more subject to noise, and if used for direclty power a board, it can create problems.

Myzhar, 2015-01-12:
If you think to use the regulated 5V to power external encoders without the necessity to use other boards, I agree with you and I say that the uNav must have a 5V regulator onboard that generates that power voltage.

The only thing that makes me thinking is the power dissipation. Do we have enough space to place a power dissipator for the linear regulator?

@katodo @rbonghi @Marnic @guiott

katodo, 2015-01-12:
Have you statistic about encoder power consumption? there are a lot of
encoder type....
Can you also measure power supply current of uNav with software running?

Mauro Soligo by Sony Xperia Z

katodo, 2015-01-12:
Ok, I've analyzed the problem and 5V are required! :)

Now, I can keep old solution with 2 LDO but I've found some better solution:
- http://www.analog.com/static/imported-files/data_sheets/ADP1864.pdf
- http://www.torex.co.jp/english/products/dcdc_converters/data/XC9503.pdf

First code can be powered with a supply up to 14V, second IC is dual-output and can be used for 3,3/5V out with a supply from 6 to 10V  
They are a more expensive solution ( about 2€ Vs. <1€ of LDO version ) but is a very efficiency....

rbonghi, 2015-01-13:
I think second IC is more convenient to save space on PCB board and we don't use second layer to put ICs.

Finally I think is important to put a 5V regulator to power (in stand-alone version) the encoders.

katodo, 2015-01-13:
Price is 1,3€/each for 100pcs on digikey, with inductor and external mosfet solution can be expensive... now I search another IC...

katodo, 2015-01-13:
Another idea, a 2-stage supply:
Vin -> 5V with:
http://www.digikey.it/product-detail/it/AP6503SP-13/AP6503SP-13DICT-ND/2767830
Less than 0,7€ and up to 23V input

5V -> 3,3V with:
http://diodes.com/datasheets/AP3417C.pdf
http://www.digikey.it/product-detail/it/AP3417CKTR-G1/AP3417CKTR-G1DICT-ND/4505308
Less than 0,3€, without diode ( SYNCHRONOUS ) with a 2,2uH inductor ( @ 1,5Mhz ), can be the perfect solution for 3,3Vout in a space-save idea...

This is a complex but professional solution and no extremely expensive... with inductor and capacitor we are below 2,5€

@katodo @rbonghi @Marnic @guiott @bobboteck can be ok?

rbonghi, 2015-01-13:
ok! :+1:

bobboteck, 2015-01-13:
> Another idea, a 2-stage supply:
> Vin -> 5V with:
> http://www.digikey.it/product-detail/it/AP6503SP-13/AP6503SP-13DICT-ND/2767830
> Less than 0,7€ and up to 23V input
>
> 5V -> 3,3V with: [...]

If there is enough space, for me is OK!!!

Marnic, 2015-01-13:
OK

katodo, 2015-01-20:
In final revision of scheme there are 2-stage switching reglator and optional LDO regulator... I consider closed this issue.

closed 2015-01-20 by katodo