officinerobotiche/uNAVPCB #3: Analog input analisys

katodo, 2015-01-13:
In uNav board there are 4 inputs reserved for analog operation.

In my idea all input must be "general porpouse" with a 0-3,3V input range and a medium high input impedance.

I've remove OA in board because pic input is also a "semi-general-porpouse" input and Operational Amplifier-related section is very conditioned by type of measure:

Measure of current in L'298 like shunt resistor require a gain depending by current and shunt resistor. For example with 2A on 0,1 Ohm resistor I need a gain of 16,5 to obtain 0-3,3 V range, but with a different driver driver like Allegro A4950 I need a 3,7 gain ( 3,5A and 0,25 Ohm resistor )... In this case OA section is not required and is only redundant.
In other bridge with hall-effect current sensor IC there are +/-5A +/-10A or bigger full-scale ( 2,5V @ 0A measure ) and a programmable OA can be required for a big accuracy on low current measure.
Also, a filter can be required...

Input for Voltage measure require a different section, I need a voltage divider and a filter depending by type of measured voltage ( input voltage, motor coil voltage etc... )

Besides, measure conditioning mus be near source of current/voltage under test... and in all case input section can't be equal and fixed...

guiott, 2015-01-13:
In the dsNav version of motor controller (see attached) there is an amplifier with a gain of 1 or 10 selectable by a soldering jumper. That's enough for a L298 h-bridge too, as I've used on Rino. Furthermore there is a minimal low-pass filter to filter out the PWM frequency. Further filtering is achieved via software procedures: using DMA of dsPIC33F ADC you can get 16 samples with no CPU intervention and compute the average after the interrupt trigger.
I agree at all that this can be redundant in many cases and the best solution is to condition the signal where it's generated, but this keep the board more general purpose.

On the other hand a simpler board, with less components, placed on the top layer only, costs less and results more attractive to most of the possible customers.

Is it possible to calculate the cost saving in the simpler solution? I think this is the best parameter to discuss on.

> In uNav board there are 4 inputs reserved for analog operation.
>
> In my idea all input must be "general porpouse" with a 0-3,3V input range and a medium high input impedance.
>
> I've remove OA in board [...]

___________________
http://www.guiott.com/

katodo, 2015-01-13:
Ok, I'm evaluating this IC:
http://ww1.microchip.com/downloads/en/DeviceDoc/22004b.pdf
http://it.farnell.com/microchip/mcp6g01t-e-ot/amp-select-gain-single-1-mhz-smd/dp/1332148?ost=1332148

Gain is pin-selectable from 1, 10 and 50 value, IC is Rail-ToR-Rail and supplies is 1,8-5V range...
IC is small but I suspect that bottom side wil be needed and filer can't be perfomed in HW... 
Voltage supply can be the same of dsPIC... ad will be the same...

During PCB update I will choose ideal version of IC ( 4pcs 1-channel or 1 IC at 4 ch ).
About 2 side version cost's, there are a lot of thing to be evaluated but in this stage I consider that, if there are some  IC on bottom side of PCB, they can be hand soldered.

@guiott @Marnic @Myzhar @rbonghi @bobboteck  can you "vote" this possible solution ( +1 or -1 )?

guiott, 2015-01-13:
Like!

> Ok, I'm evaluating this IC:
> http://ww1.microchip.com/downloads/en/DeviceDoc/22004b.pdf <http://ww1.microchip.com/downloads/en/DeviceDoc/22004b.pdf> [...]

___________________
http://www.guiott.com/ <http://www.guiott.com/>

Marnic, 2015-01-13:
OK
1++

Data: Tue, 13 Jan 2015 01:14:45 -0800
Oggetto: Re: [uNAVPCB] Analog input analisys (#3)

katodo, 2015-01-20:
in final revison of scheme I've added OA with configurable GAIN using MCU I/O... I consider closed this issue

closed 2015-01-20 by katodo