officinerobotiche/uNAVPCB #4: Component exchange for miniaturization

katodo, 2015-01-14:
can be exchanged this component?
- Actual 24AA01 EEPROM with this  AT24C16D-STUM-T 16Kbit in SOT23 case?
  http://www.digikey.it/product-detail/it/AT24C16D-STUM-T/AT24C16D-STUM-TCT-ND/5055821

- Actual 10Mhz xtal with this:
  http://www.digikey.it/product-detail/it/NX5032GA-10.000000MHZ-LN-CD-1/644-1033-1-ND/1128905

@rbonghi @guiott @Cyb3rn0id @Marnic @bobboteck

bobboteck, 2015-01-15:
I think there is no problem, you can talk to @guiott that developed the code to control the eeprom

katodo, 2015-01-15:
EEprom is last IC that can be changed but is a possibility.

Cordiali saluti
                        Mauro Soligo

Katodo  di Mauro Soligo
Via Molino Nuovo, 31 35012 Camposampiero PD Italy
P.IVA 04161520285  C.F. SLGMRA80L18B563Y
Cell: +39 3462137146  FAX: +39 0498591588  e-mail mauro.soligo@katodo.com

rbonghi, 2015-01-15:
I'm agree to increase free area space on unav pcb board. :+1:

katodo, 2015-01-19:
Only XTAL was changed.

closed 2015-01-19 by katodo