officinerobotiche/uNAVPCB #14: Discussion of board

Russ76, 2017-03-07:
Do you have somewhere posted a comprehensive discussion of what you are trying to accomplish with this project, and how this board(s) will be superior to other alternatives already on the market? Will it control large motors, like 25 amps? Is the ROS software available now for use of this board?

Myzhar, 2017-03-07:
@Russ76 you are putting good questions. 

We are still trying to understand if "reinventing the wheel" is a good road to follow. 

The starting idea was the one to create by ourself a tiny dual channels motor driver (4cm x 4cm)  with logic and power separated. This module (uNAVPCB) cannot directly drive any motor because it only generates two PWM signals to be sent to an external H-Bridge, controlling speed (dual channels QEI encoders are supported) and torque, if you can provide analog current sensing. 

I can reply to the second of your question: YES, you can drive large motors, only if you use the right H-Bridge for that kind of motors. The uNAV can theoretically drive any kind of motor...

About ROS... yes, we have ROS drivers:  https://github.com/officinerobotiche/ros_orbus_interface
Here you can find the also the firmware for the board: https://github.com/officinerobotiche/uNAV.X

Now the reply to the first question... it is a phylosophical question because analizing the market I can say that there is not the needing for a new motor controller board. You can find a lot of good motor drivers on the market, with a good ROS support, RoboteQ's (e.g. SDC2160) and Ion Motion Control's (RoboClaws) are good examples. Sometimes the market must not be a limitation, a new project can also be a good way to make practice with hardware and to study new software algorithms.
To finish this long reply: if I need to go on the market with a new "robot" I will not create a new motor controller board, but I will use something available off the shelf to not waste time and money; if I want to make a bit of research in the field of motor control I will create my own board trying to understand what I can do better than other product on the market (i.e. speed and torque control on a dual PID loop).

I hope to have replied to all of your question, please let me know what you think about.

Walter

Russ76, 2017-03-07:
Thanks for that, Walter! Your board is very small, that is good. And I'm sure the research and project operation was educational for several of you. Building a robot at this time is an educational project and not easy; there is so much to learn. Just learning ROS itself is challenging.
For motor control I'm using a Sabertooth and Kangaroo. Just installed encoders last week. We'll see how it works. Please let us know when your uNav is market ready!

Myzhar, 2017-03-07:
@Russ76 Thank you

closed 2017-03-16 by Russ76